IndaloTech/ramses_esp #1: CR LF characters between messages

brucemiranda, 2024-01-03:
On the ESP device, the very first message is not appearing on a new line because of a missing CR LF between the debug messages of the bootloader and the first message.

Also all subsequent messages have CR CR LF printed between two messages rather than just CR LF..So depending on the text editor used, shows a blank line between messages.

zxdavb, 2024-01-13:
For reference, this is what I see:
```
Tx:     b'RQ --- 18:000730 01:145038 --:------ 2E04 001 FF \r\n'
Rx:     b'RQ --- 18:000730 01:145038 --:------ 2E04 001 FF \r\n'
Rx: b' 000 RQ --- 18:017804 01:145038 --:------ 2E04 001 FF\r\r\n'
Rx: b' \r\n'
Rx: b' 053 RP --- 01:145038 18:017804 --:------ 2E04 008 00FFFFFFFFFFFF00\r\r\n'
```

... this is what I am expecting (i.e. behaviour of HGI80):
```
Tx:     b'RQ --- 18:000730 01:145038 --:------ 2E04 001 FF \r\n'
Rx: b'000 RQ --- 18:017804 01:145038 --:------ 2E04 001 FF\r\n'
Rx: b'053 RP --- 01:145038 18:017804 --:------ 2E04 008 00FFFFFFFFFFFF00\r\n'
```

(as I write this - I see I need to double-check: I will remove this line when I have done so)

ghoti57, 2024-03-22:
Fixed in v0.4.0

closed 2024-03-22 by ghoti57