IndaloTech/ramses_esp #12: Without sntp frequent mqtt disconnects

sandervandegeijn, 2024-12-31:
If you don't have SNTP configured you'll see a lot of MQTT disconnects. Handling could be better I think, either make SNTP a requirement, provide sensible defaults or prevent the disconnects. Running the latest stable compiled to an ESP32-S3.

zxdavb, 2025-02-18:
@ghoti57 This is causing issues with HA as well.

ghoti57, 2025-02-18:
SNTP specifically state that embedded software should not include default links to general

ghoti57, 2025-02-18:
RFC4340 specifically states <https://datatracker.ietf.org/doc/html/rfc4330> Section 10 Para 5.

> If a firmware default server IP address is provided, it MUST be a
       server operated by the manufacturer or seller of the device or
       another server, but only with the operator's permission.

I am not in a position to provide a server.

ghoti57, 2025-02-18:
Are the HA issues associated with Issue 4 <https://github.com/IndaloTech/ramses_esp/issues/13>

zxdavb, 2025-02-19:
The issue is that upper layers of **ramses_rf** will 'expire' a packet that is older that a certain amount of time - e.g. 15 mins * 2.5, if it knows the packet comes every 15 mins.

If the datetime is not set on **ramses_esp** by SNTP, then (IIRC) it defaults to 1970-01-01T00:00:00, and goes up from there...

Of course, all this packets will be 'expired' immediately.

The other option is to simply not forward packets to MQTT until the SNTP settings are configured, and throw a warning instead.

sandervandegeijn, 2025-02-19:
That would be cleaner if it's really necessary. At least it's an expliciet failure.

ghoti57, 2025-02-19:
I'll try to publish a status entry in mqtt if SNTP is not configured. 

 I'll also try to prevent publishing RX messages.

 The slightly trickier scenario is SNTP has been configured but the reset the code currently requires hasn't happened.  I need to work out when/how to start/restart SNTP

closed 2026-10-31 by ghoti57