IndaloTech/ramses_esp #13: Configuration settings to use with HA MQTT Broker Add-On

brucemiranda, 2025-01-03:
When using the MQTT Broker Add-On in HA, the MQTT Broker is run within a separate container. So a lot of configuration settings of the Broker change when HA is restarted.

How do you configure the ESP device and ramses_cc to use the MQTT Broker Add-On in HA?

ghoti57, 2025-02-18:
I don't use MQTT Broker, I use a eclipse:MQTT in a separate container on my NAS.

@zxdavb mentioned it to me and so I pointed it out to potential users but I didn't test it.
Anybody who has worked out the answer to this please respond

brucemiranda, 2025-02-19:
On HAOS atleast the MQTT broker IP address is the same as the HA Host itself.  All the other options stay as the default. e.g. the port numbers etc. 
You then have to create an HA user and password via Settings, People, Users. The HA MQTT Broker does not allow anonymous logins. 
You use these 3 items to update the MQTT configuration settings on the ESP device.

mqtt user _ha_mqtt_user_
mqtt password _ha_mqtt_userr_password_
mqtt broker mqtt://_ha_host_ip_:1883