IndaloTech/ramses_esp #21: Backtrace on fresh Ramses ESP

danielvandenberg95, 2025-05-11:
I just plugged in my new Ramses ESP, windows, putty, COM3 at 115200. I'm repeatedly getting:
```
Backtrace: 0x40379986:0x3FC9DB30 0x40377811:0x3FC9DB50 0x400559DD:0x3FCB25F0 0x4037EE27:0x3FCB2600 0x4037E7F6:0x3FCB2620 0x42019299:0x3FCB2660 0x420192D2:0x3FCB2690 0x4200CE61:0x3FCB26C0 0x4200CFB5:0x3FCB2710 0x4200CFFA:0x3FCB2750 0x4200A8DB:0x3FCB2770 0x4200ADCF:0x3FCB2790 0x4200A212:0x3FCB27B0 0x4037EA7A:0x3FCB27D0

Backtrace: 0x40379986:0x3FC9DB30 0x40377811:0x3FC9DB50 0x400559DD:0x3FCB25F0 0x4037EE27:0x3FCB2600 0x4037E7F6:0x3FCB2620 0x42019299:0x3FCB2660 0x420192D2:0x3FCB2690 0x4200CE61:0x3FCB26C0 0x4200CFB5:0x3FCB2710 0x4200CFFA:0x3FCB2750 0x4200A8DB:0x3FCB2770 0x4200ADCF:0x3FCB2790 0x4200A212:0x3FCB27B0 0x4037EA7A:0x3FCB27D0
```

Typing help+enter does seem to work, but I'm assuming my device keeps restarting. I'll try configuring it to see if the issue persists, but I would like to know if this can hurt. It does seem poor UX in any case.

Do I need to re-flash it?

danielvandenberg95, 2025-05-11:
Unfortunately after configuring my wifi (and checking about 5 times), I can confirm it doesn't work:

```
# reset
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b2c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x48c
load:0x403c9700,len:0x4
load:0x403c9704,len:0x990
load:0x403cc700,len:0x25a0
entry 0x403c9858
# ramses_esp 0.4.9
# Attempting to connect to SSID:van den Berg

Backtrace: 0x40379986:0x3FC9DB30 0x40377811:0x3FC9DB50 0x40376895:0x3FCB26E0 0x40378165:0x3FCB2700 0x420B11FA:0x3FCB2720 0x4201B845:0x3FCB2740 0x4201920E:0x3FCB2760 0x420192C5:0x3FCB2790 0x4200CE61:0x3FCB27C0 0x4200CFB5:0x3FCB2810 0x4200CFFA:0x3FCB2850 0x4200D0ED:0x3FCB2870 0x4200AD99:0x3FCB28C0 0x4200A20C:0x3FCB28E0 0x4037EA7A:0x3FCB2900

Backtrace: 0x40379986:0x3FC9DB30 0x40377811:0x3FC9DB50 0x400559DD:0x3FCB26F0 0x4037EE27:0x3FCB2700 0x4037E7B2:0x3FCB2720 0x42019299:0x3FCB2760 0x420192D2:0x3FCB2790 0x4200CE61:0x3FCB27C0 0x4200CFB5:0x3FCB2810 0x4200CFFA:0x3FCB2850 0x4200D0ED:0x3FCB2870 0x4200AD99:0x3FCB28C0 0x4200A20C:0x3FCB28E0 0x4037EA7A:0x3FCB2900

# Failed to connect to SSID:van den Berg

Backtrace: 0x40379986:0x3FC9DB30 0x40377811:0x3FC9DB50 0x400559DD:0x3FCB26F0 0x4037EE27:0x3FCB2700 0x4037E7F6:0x3FCB2720 0x42019299:0x3FCB2760 0x420192D2:0x3FCB2790 0x4200CE61:0x3FCB27C0 0x4200CFB5:0x3FCB2810 0x4200CFFA:0x3FCB2850 0x4200D0ED:0x3FCB2870 0x4200AD99:0x3FCB28C0 0x4200A20C:0x3FCB28E0 0x4037EA7A:0x3FCB2900
```

danielvandenberg95, 2025-05-11:
I can confirm that it does not seem to work. I've connected it to MQTT, and all I'm getting is (read bottom to top, I issued a RESET via serial after message 2):
```
Message 7 received on RAMSES/GATEWAY/18:054684/info/version at 10:35:
0.4.9
QoS: 0 - Retain: false
Message 6 received on RAMSES/GATEWAY/18:054684/info/firmware at 10:35:
ramses_esp
QoS: 0 - Retain: false
Message 5 received on RAMSES/GATEWAY/18:054684/cmd/cmd at 10:35:
QoS: 0 - Retain: false
Message 4 received on RAMSES/GATEWAY/18:054684 at 10:35:
online
QoS: 0 - Retain: false
Message 3 received on RAMSES/GATEWAY/18:054684 at 10:35:
offline
QoS: 0 - Retain: false
Message 2 received on RAMSES/GATEWAY/18:054684/info/version at 10:33:
0.4.9
QoS: 0 - Retain: true
Message 1 received on RAMSES/GATEWAY/18:054684/info/firmware at 10:33:
ramses_esp
QoS: 0 - Retain: true
Message 0 received on RAMSES/GATEWAY/18:054684 at 10:33:
online
QoS: 0 - Retain: true
```

I've tried ota force. Same result.

danielvandenberg95, 2025-05-11:
After building manually:
```
Backtrace: 0x403791CE:0x3FC9FE00 0x40377975:0x3FC9FE20 0x400559DD:0x3FCB6960 0x4037F6BA:0x3FCB6970 0x4037F081:0x3FCB6990 0x42024725:0x3FCB69D0 0x4202475E:0x3FCB6A10 0x4200ECE1:0x3FCB6A40 0x4200EE2D:0x3FCB6A90 0x4200EE72:0x3FCB6AD0 0x4200EF61:0x3FCB6AF0 0x4200CC75:0x3FCB6B40 0x4200BE30:0x3FCB6B60 0x4037F31A:0x3FCB6B80
--- 0x403791ce: esp_crosscore_isr at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_system/crosscore_int.c:73
--- 0x40377975: _xt_lowint1 at C:/Users/danie/esp/v5.4.1/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x400559dd: _xtos_set_intlevel in ROM
--- 0x4037f6ba: vPortClearInterruptMaskFromISR at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:560
---  (inlined by) vPortExitCritical at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:514      
--- 0x4037f081: xQueueReceive at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1632
--- 0x42024725: spi_device_get_trans_result at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_driver_spi/src/gpspi/spi_master.c:1257
--- 0x4202475e: spi_device_transmit at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_driver_spi/src/gpspi/spi_master.c:1284
--- 0x4200ece1: spi_write_byte at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:68
--- 0x4200ee2d: cc_strobe at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:121
--- 0x4200ee72: cc_enter_rx_mode at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:146 (discriminator 1)
--- 0x4200ef61: cc_init at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:211
--- 0x4200cc75: frame_init at C:/Users/danie/Documents/Workspaces/ramses_esp/components/frame/frame.c:627
--- 0x4200be30: Radio_Task at C:/Users/danie/Documents/Workspaces/ramses_esp/main/radio.c:34
--- 0x4037f31a: vPortTaskWrapper at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

Backtrace: 0x403791CE:0x3FC9FE00 0x40377975:0x3FC9FE20 0x400559DD:0x3FCB6960 0x4037F6BA:0x3FCB6970 0x4037F081:0x3FCB6990 0x42024725:0x3FCB69D0 0x4202475E:0x3FCB6A10 0x4200ECE1:0x3FCB6A40 0x4200EE2D:0x3FCB6A90 0x4200EE72:0x3FCB6AD0 0x4200EF61:0x3FCB6AF0 0x4200CC75:0x3FCB6B40 0x4200BE30:0x3FCB6B60 0x4037F31A:0x3FCB6B80
--- 0x403791ce: esp_crosscore_isr at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_system/crosscore_int.c:73
--- 0x40377975: _xt_lowint1 at C:/Users/danie/esp/v5.4.1/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x400559dd: _xtos_set_intlevel in ROM
--- 0x4037f6ba: vPortClearInterruptMaskFromISR at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:560
---  (inlined by) vPortExitCritical at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:514      
--- 0x4037f081: xQueueReceive at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1632
--- 0x42024725: spi_device_get_trans_result at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_driver_spi/src/gpspi/spi_master.c:1257       
--- 0x4202475e: spi_device_transmit at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_driver_spi/src/gpspi/spi_master.c:1284
--- 0x4200ece1: spi_write_byte at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:68
--- 0x4200ee2d: cc_strobe at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:121
--- 0x4200ee72: cc_enter_rx_mode at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:146 (discriminator 1)
--- 0x4200ef61: cc_init at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:211
--- 0x4200cc75: frame_init at C:/Users/danie/Documents/Workspaces/ramses_esp/components/frame/frame.c:627
--- 0x4200be30: Radio_Task at C:/Users/danie/Documents/Workspaces/ramses_esp/main/radio.c:34
--- 0x4037f31a: vPortTaskWrapper at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

```

danielvandenberg95, 2025-05-11:
Adding logs:
```
void cc_enter_rx_mode(void) {
  printf("# CC1101 RX mode\n");
  while ( CC_STATE( cc_strobe( CC_SIDLE ) ) != CC_STATE_IDLE ){}
  printf("# CC1101 RX idle\n");
  printf("# Disable GDO0: %02x\n", cc_write( CC_IOCFG0, 0x2E ));      // GDO0 not needed
  printf("# Async RX mode: %02x\n", cc_write( CC_PKTCTRL0, 0x32 ));	// Asynchronous, infinite packet
  printf("# Strobe CC_SFRX: %02x\n", cc_write( CC_SFRX, 0x00 ));	// RX Fifo flush
  printf("Waiting for RX\n");
  uint8_t state = CC_STATE_IDLE;
  while ( state != CC_STATE_RX ){
    uint8_t new_state = CC_STATE( cc_strobe( CC_SRX ) );
    if ( new_state != state ) {
      state = new_state;
      printf("State changed to %02x\n", state );
    }
  }
  printf("# CC1101 now in RX mode\n");
}
```
gives me:
```
CC1101 init
# CC1101: 0x3fcbb0a0       
# CC1101 pre RX: 0x3fcbb0a0
# CC1101 RX mode
# CC1101 RX idle
# Disable GDO0: 0f
# Async RX mode: 0f
# Strobe CC_SFRX: 0f
Waiting for RX
State changed to 40
State changed to 50
State changed to 00
# Attempting to connect to SSID:van den Berg
# Connected to SSID:van den Berg
# MQTT: Connecting to mqtt://192.168.0.64:1883
# MQTT: Connected

Backtrace: 0x403791CE:0x3FC9FE00 0x40377975:0x3FC9FE20 0x400559DD:0x3FCB6960 0x4037F6BA:0x3FCB6970 0x4037F081:0x3FCB6990 0x420247D1:0x3FCB69D0 0x4202480A:0x3FCB6A10 0x4200ED11:0x3FCB6A40 0x4200EE5D:0x3FCB6A90 0x4200EEE4:0x3FCB6AD0 0x4200F00F:0x3FCB6AF0 0x4200CCA5:0x3FCB6B40 0x4200BE60:0x3FCB6B60 0x4037F31A:0x3FCB6B80
--- 0x403791ce: esp_crosscore_isr at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_system/crosscore_int.c:73
--- 0x40377975: _xt_lowint1 at C:/Users/danie/esp/v5.4.1/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x400559dd: _xtos_set_intlevel in ROM
--- 0x4037f6ba: vPortClearInterruptMaskFromISR at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:560
---  (inlined by) vPortExitCritical at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:514
--- 0x4037f081: xQueueReceive at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1632
--- 0x420247d1: spi_device_get_trans_result at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_driver_spi/src/gpspi/spi_master.c:1257
--- 0x4202480a: spi_device_transmit at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_driver_spi/src/gpspi/spi_master.c:1284
--- 0x4200ed11: spi_write_byte at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:68
--- 0x4200ee5d: cc_strobe at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:121
--- 0x4200eee4: cc_enter_rx_mode at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:149
--- 0x4200f00f: cc_init at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:226
--- 0x4200cca5: frame_init at C:/Users/danie/Documents/Workspaces/ramses_esp/components/frame/frame.c:627
--- 0x4200be60: Radio_Task at C:/Users/danie/Documents/Workspaces/ramses_esp/main/radio.c:34
--- 0x4037f31a: vPortTaskWrapper at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

Backtrace: 0x403791CE:0x3FC9FE00 0x40377975:0x3FC9FE20 0x40379536:0x3FCB69A0 0x420246C6:0x3FCB69D0 0x420247FD:0x3FCB6A10 0x4200ED11:0x3FCB6A40 0x4200EE5D:0x3FCB6A90 0x4200EEE4:0x3FCB6AD0 0x4200F00F:0x3FCB6AF0 0x4200CCA5:0x3FCB6B40 0x4200BE60:0x3FCB6B60 0x4037F31A:0x3FCB6B80        
--- 0x403791ce: esp_crosscore_isr at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_system/crosscore_int.c:73
--- 0x40377975: _xt_lowint1 at C:/Users/danie/esp/v5.4.1/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x40379536: check_trans_valid at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_driver_spi/src/gpspi/spi_master.c:1043 (discriminator 4)
--- 0x420246c6: spi_device_queue_trans at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_driver_spi/src/gpspi/spi_master.c:1196
--- 0x420247fd: spi_device_transmit at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_driver_spi/src/gpspi/spi_master.c:1279
--- 0x4200ed11: spi_write_byte at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:68
--- 0x4200ee5d: cc_strobe at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:121
--- 0x4200eee4: cc_enter_rx_mode at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:149
--- 0x4200f00f: cc_init at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:226
--- 0x4200cca5: frame_init at C:/Users/danie/Documents/Workspaces/ramses_esp/components/frame/frame.c:627
--- 0x4200be60: Radio_Task at C:/Users/danie/Documents/Workspaces/ramses_esp/main/radio.c:34
--- 0x4037f31a: vPortTaskWrapper at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
[... 1 lines cut ...]
Backtrace: 0x403791CE:0x3FC9FE00 0x40377975:0x3FC9FE20 0x400559DD:0x3FCB6920 0x4037F6BA:0x3FCB6930 0x40376BD0:0x3FCB6950 0x40379981:0x3FCB6970 0x420B39FA:0x3FCB6990 0x4201CFD1:0x3FCB69B0 0x42024742:0x3FCB69D0 0x420247FD:0x3FCB6A10 0x4200ED11:0x3FCB6A40 0x4200EE5D:0x3FCB6A90 0x4200EEE4:0x3FCB6AD0 0x4200F00F:0x3FCB6AF0 0x4200CCA5:0x3FCB6B40 0x4200BE60:0x3FCB6B60 0x4037F31A:0x3FCB6B80
--- 0x403791ce: esp_crosscore_isr at C:/Users/danie/esp/v5.4.1/esp-idf/components/esp_system/crosscore_int.c:73
--- 0x40377975: _xt_lowint1 at C:/Users/danie/esp/v5.4.1/esp-idf/components/xtensa/xtensa_vectors.S:1240
[... cut ...]
--- 0x4200eee4: cc_enter_rx_mode at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:149
--- 0x4200f00f: cc_init at C:/Users/danie/Documents/Workspaces/ramses_esp/components/cc1101/cc1101.c:226
--- 0x4200cca5: frame_init at C:/Users/danie/Documents/Workspaces/ramses_esp/components/frame/frame.c:627
--- 0x4200be60: Radio_Task at C:/Users/danie/Documents/Workspaces/ramses_esp/main/radio.c:34
--- 0x4037f31a: vPortTaskWrapper at C:/Users/danie/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

```

So the state goes 40, 50, 0 and then stays idle.

danielvandenberg95, 2025-05-11:
Perhaps more relevant info:
```
void cc_enter_rx_mode(void) {
  printf("# CC1101 RX mode\n");
  while ( CC_STATE( cc_strobe( CC_SIDLE ) ) != CC_STATE_IDLE ){}
  printf("# CC1101 RX idle\n");
  printf("# Disable GDO0: %02x\n", cc_write( CC_IOCFG0, 0x2E ));      // GDO0 not needed
  printf("# Async RX mode: %02x\n", cc_write( CC_PKTCTRL0, 0x32 ));	// Asynchronous, infinite packet
  vTaskDelay(pdMS_TO_TICKS(1));
  printf("# Strobe CC_SFRX: %02x\n", cc_strobe(CC_SFRX));  // Flush RX FIFO
  vTaskDelay(pdMS_TO_TICKS(1));
  printf("Waiting for RX\n");
  uint8_t state = CC_STATE_IDLE;
  bool was_not_idle = false;
  while ( CC_STATE( state) != CC_STATE_RX ){
    uint8_t new_state = cc_strobe( CC_SRX );
    if ( new_state != state ) {
      state = new_state;
      printf("State changed to %02x\n", state );
    }
    if (CC_STATE( state) != CC_STATE_IDLE && !was_not_idle) {
      was_not_idle = true;
    }
    if (CC_STATE( state) == CC_STATE_IDLE && was_not_idle) {
      printf("CC1101 RX idle\n");
      printf("Unexpectedly returned to IDLE state during RX mode\n");
      printf("Diagnostics:\n");
      printf("  Current state: %02x\n", state);
      printf("  MARCSTATE: %02x\n", cc_read(CC_MARCSTATE));
      printf("  RXBYTES: %02x\n", cc_read(CC_RXBYTES));
      printf("  PKTSTATUS: %02x\n", cc_read(CC_PKTSTATUS));
      vTaskDelay(pdMS_TO_TICKS(20000)); // Delay for 20 seconds
      abort(); // Terminate the program immediately
    }
  }
  printf("# CC1101 now in RX mode\n");
}
```
```
# ramses_esp 0.5.2
CC1101 init
# CC1101: 0x3fcbb384
# CC1101 pre RX: 0x3fcbb384
# CC1101 RX mode
# CC1101 RX idle
# Disable GDO0: 0f
# Async RX mode: 0f
# Strobe CC_SFRX: 0f
Waiting for RX
State changed to 0f
State changed to 4f
State changed to 5e
State changed to 00
CC1101 RX idle
Unexpectedly returned to IDLE state during RX mode
Diagnostics:
  Current state: 00
  MARCSTATE: 00
  RXBYTES: 00
  PKTSTATUS: 00
```

danielvandenberg95, 2025-05-11:
By commenting out cc_enter_rx_mode in cc_init I can get it to go into state "1d" for about 5 seconds, after which it restarts (or so) and goes to state 00. I think it is being switched by the main "frame" loop, but did not dig that deep yet.

```
# ramses_esp 0.5.2
CC1101 init
# CC1101: 0x3fcbb390
# CC1101 pre RX: 0x3fcbb390
# CC1101 RX mode
# CC1101 RX idle
# Disable GDO0: 0f
# Async RX mode: 0f
# Strobe CC_SFRX: 0f
Waiting for RX
Strobe CC_SRX: 0f
State changed to 4f
State changed to 1d
# CC1101 now in RX mode
--- There's about a 5 second delay here somewhere.
# Attempting to connect to SSID:van den Berg
# Connected to SSID:van den Berg
# MQTT: Connecting to mqtt://192.168.0.64:1883
# MQTT: Connected
# CC1101 RX mode
# CC1101 RX idle
# Disable GDO0: 0f
# Async RX mode: 0f
# Strobe CC_SFRX: 0f
Waiting for RX
Strobe CC_SRX: 0f
State changed to 4f
State changed to 5f
State changed to 00
CC1101 RX idle
# CC1101 now in RX mode
```

danielvandenberg95, 2025-05-11:
I'm starting to think it's a power issue on my board. The marcstate suddenly also goes from 08 to 00, which shouldn't ever happen. Any read afterwards gives 00. That would be consistent with the CC1101 not responding.
```
void cc_enter_rx_mode(void) {
  printf("# CC1101 RX mode\n");
  while ( CC_STATE( cc_strobe( CC_SIDLE ) ) != CC_STATE_IDLE ){}
  printf("# CC1101 RX idle\n");
  printf("# Disable GDO0: %02x\n", cc_write( CC_IOCFG0, 0x2E ));      // GDO0 not needed
  printf("Marcstate: %02x\n", cc_read( CC_MARCSTATE )); // Check current state
  printf("# Async RX mode: %02x\n", cc_write( CC_PKTCTRL0, 0x32 ));	// Asynchronous, infinite packet
  vTaskDelay(pdMS_TO_TICKS(1));
  printf("# Strobe CC_SFRX: %02x\n", cc_strobe(CC_SFRX));  // Flush RX FIFO
  vTaskDelay(pdMS_TO_TICKS(1));
  printf("Waiting for RX\n");
  uint8_t state = cc_strobe(CC_SRX);
  printf("Strobe CC_SRX: %02x\n", state);
  for (int i=0; i<10; i++) {
    printf("  MARCSTATE: %02x\n", cc_read(CC_MARCSTATE));
  }
  while ( CC_STATE( state) != CC_STATE_RX ){
    uint8_t new_state = cc_strobe( CC_SRX );
    if ( new_state != state ) {
      state = new_state;
      printf("State changed to %02x\n", state );
    }
    if (CC_STATE( state) == CC_STATE_IDLE) {
      printf("Unexpectedly returned to IDLE state during RX mode\n");
      printf("Diagnostics:\n");
      printf("  Current state: %02x\n", state);
      printf("  MARCSTATE: %02x\n", cc_read(CC_MARCSTATE));
      printf("  RXBYTES: %02x\n", cc_read(CC_RXBYTES));
      printf("  PKTSTATUS: %02x\n", cc_read(CC_PKTSTATUS));
      printf("  RSSI: %02x\n", cc_read(CC_RSSI));
      printf("  LQI: %02x\n", cc_read(CC_LQI));
      printf("  RXFIFO: %02x\n", cc_read(CC_RXFIFO));
      printf("  TXFIFO: %02x\n", cc_read(CC_TXFIFO));
      printf("  PKTCTRL0: %02x\n", cc_read(CC_PKTCTRL0));
      printf("  PKTCTRL1: %02x\n", cc_read(CC_PKTCTRL1));
      vTaskDelay(pdMS_TO_TICKS(5 * 1000)); // Delay
      
      printf("Trying to enter RX mode again\n");
      vTaskDelay(pdMS_TO_TICKS(1));
      state = cc_strobe(CC_SRX);
      printf("Strobe CC_SRX: %02x\n", state);
      printf("  MARCSTATE: %02x\n", cc_read(CC_MARCSTATE));
    }
  }
  printf("# CC1101 now in RX mode\n");
}
```
```
# ramses_esp 0.5.2
CC1101 init
# CC1101: 0x3fcbb0bc
# CC1101 pre RX: 0x3fcbb0bc
# CC1101 RX mode
# CC1101 RX idle
# Disable GDO0: 0f
Marcstate: 01
# Async RX mode: 0f
# Strobe CC_SFRX: 0f
Waiting for RX
Strobe CC_SRX: 0f
  MARCSTATE: 08 << This time it was here
  MARCSTATE: 00
  MARCSTATE: 00 << Sometimes this one is also still 08
  MARCSTATE: 00
  MARCSTATE: 00
  MARCSTATE: 00
  MARCSTATE: 00
  MARCSTATE: 00
  MARCSTATE: 00
  MARCSTATE: 00
State changed to 00
Unexpectedly returned to IDLE state during RX mode
Diagnostics:
  Current state: 00
  MARCSTATE: 00
  RXBYTES: 00
  PKTSTATUS: 00
  RSSI: 00
  LQI: 00
  RXFIFO: 00
  TXFIFO: 00
  PKTCTRL0: 00
  PKTCTRL1: 00 << This should never be 00...
```
From here on it repeats but only returns 00, also for strobes. What can I still try/do?

danielvandenberg95, 2025-05-13:
A bit more progress on localizing where the issue begins:
```
void cc_init(void) {
  uint8_t param[CC_PARAM_MAX];
  uint8_t i,len;

  esp_log_level_set(TAG, CONFIG_CC_LOG_LEVEL );

  cc_spi_reset();
  cc_spi_init();

  cc_strobe(CC_SRES);
  //cc_strobe(CC_SCAL);

  len = cc_cfg_get( 0, param, sizeof(param) );
  for ( i=0 ; i<len ; i++ )
    cc_write( i, param[i] );

  cc_write( CC_FIFOTHR, ( param[CC_FIFOTHR]&0xF0 )+14 );	  // TX Fifo Threshold 5

  len = cc_pa_get( param );
  for ( i=0 ; i<len ; i++ )
    cc_write( CC_PATABLE, param[i]);

  printf("RSSI pre RX: %02x LQI: %02x\n", cc_read(CC_RSSI), cc_read(CC_LQI));
  printf("RX\n");
  cc_strobe( CC_SRX );
  printf("MARCSTATE: %02x\n", cc_read(CC_MARCSTATE));
  for (int i=0; i<10; i++) {
    printf("RSSI: %02x LQI: %02x\n", cc_read(CC_RSSI), cc_read(CC_LQI));
    vTaskDelay(pdMS_TO_TICKS(500));
  }
  printf("MARCSTATE: %02x\n", cc_read(CC_MARCSTATE));
  vTaskDelay(pdMS_TO_TICKS(5000));

  cc_enter_rx_mode();
}
```
gives me this:
```
# ramses_esp 0.5.2
RSSI pre RX: 80 LQI: 80
RX
MARCSTATE: 08
RSSI: 80 LQI: 80
# Attempting to connect to SSID:van den Berg
RSSI: 00 LQI: 00
RSSI: 00 LQI: 00
RSSI: 00 LQI: 00
RSSI: 00 LQI: 00
RSSI: 00 LQI: 00
RSSI: 00 LQI: 00
RSSI: 00 LQI: 00
# Connected to SSID:van den Berg
# MQTT: Connecting to mqtt://192.168.0.64:1883
RSSI: 00 LQI: 00
RSSI: 00 LQI: 00
MARCSTATE: 00
# MQTT: Connected
```
which to me means "SPI crash". (Marcstate 00 is weird.)

Commenting out
```
  len = cc_cfg_get( 0, param, sizeof(param) );
  for ( i=0 ; i<len ; i++ )
    cc_write( i, param[i] );

  cc_write( CC_FIFOTHR, ( param[CC_FIFOTHR]&0xF0 )+14 );	  // TX Fifo Threshold 5

  len = cc_pa_get( param );
  for ( i=0 ; i<len ; i++ )
    cc_write( CC_PATABLE, param[i]);
```
gives me:
```
# ramses_esp 0.5.2
RSSI pre RX: 80 LQI: 63
RX
MARCSTATE: 00
RSSI: 00 LQI: 00
# Attempting to connect to SSID:van den Berg
RSSI: dd LQI: 7f
RSSI: dd LQI: 7f
RSSI: dd LQI: 7f
RSSI: dd LQI: 7f
RSSI: dd LQI: 7f
RSSI: dd LQI: 7f
RSSI: dd LQI: 7f
RSSI: dd LQI: 7f
# Connected to SSID:van den Berg
# MQTT: Connecting to mqtt://192.168.0.64:1883
# MQTT: Connected
RSSI: dd LQI: 7f
MARCSTATE: 01
```
which to me means "packet received but with incorrect CRC, returned to marcstate 01, idle."

Enabling
```
  len = cc_cfg_get( 0, param, sizeof(param) );
  for ( i=0 ; i<len ; i++ )
    cc_write( i, param[i] );
```
again makes it return RSSI and MARCSTATE 00.

Instead enabling 
```
  cc_write( CC_FIFOTHR, ( param[CC_FIFOTHR]&0xF0 )+14 );	  // TX Fifo Threshold 5

  len = cc_pa_get( param );
  for ( i=0 ; i<len ; i++ )
    cc_write( CC_PATABLE, param[i]);
```
makes it return RSSI: d5 LQI: 7f and MARCSTATE 0x11 (overflow, that would make sense as now "infinite length package" has not benen enabled.)

So one of the parameters being written is causing this.

danielvandenberg95, 2025-05-13:
```
  len = cc_cfg_get( 0, param, sizeof(param) );
  for ( i=0 ; i<len ; i++ ) {
    switch (i){
      case CC_PKTCTRL0:
      break;
      default:
      cc_write( i, param[i] );
    }
  }
```
Skipping setting PKTCTRL0 but writing all other params gives me:
```
RSSI pre RX: 80 LQI: 00
RX
MARCSTATE: 08
RSSI: 80 LQI: 00
# Attempting to connect to SSID:van den Berg
RSSI: db LQI: 00
RSSI: db LQI: 00
# Connected to SSID:van den Berg
# MQTT: Connecting to mqtt://192.168.0.64:1883
# MQTT: Connected
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
RSSI: db LQI: 00
MARCSTATE: 01
```
So that transitions to idle again. Something with the output buffers overflowing perhaps?

Wait, I'm confused now.

[image]

0x31, that's 00011111, right? Let's check the docs for PKTCTRL0:

[image]

Bit 0:1 being 11 is reserved? Shouldn't this be 0x76 or so?

danielvandenberg95, 2025-05-14:
I found the issue. The top printed part, the hole is too small, causing the antenna not to be seated properly! Removing the cover causes it to work. Adding it again makes it stop working, removing it makes it work again. 😡

Well, I found the issue and now seem to have a working RAMSES ESP. I'd heavily recommend altering the design slightly though.

danielvandenberg95, 2025-05-15:
Overnight it stopped working again. I'm suspecting a bad connection on the PCB to the antenna.

ghoti57, 2025-06-25:
TLDR

1. Anything later than v0.4.9 of ramses_esp is work in progress and should not be used
2. The instructions require the specific version of the espressif IDE because there is a known bug in that version which can easily be avoided by reverting a single commit.  I have not yet been able to verify later versions of the IDE with the RAMSES_ESP code.

closed 2026-10-31 by ghoti57